pimoroni/clean-shutdown #6: Proposal:  Minimum uptime

dbbnrl, 2017-07-14:
I've been experimenting with the Zero LIPO and clean-shutdown and run into the following scenario:

Sometimes I want to power my Pi Zero through USB power, not the LIPO connector.  The Zero LIPO supports this just fine, but in conjunction with clean-shutdown yields a problem, since without the battery connected the low-battery GPIO fires, and clean-shutdown almost immediately turns the system back off.

My proposed solution is to add a system uptime check to clean-shutdown that will cause it not to act until the system has been up for less than min_uptime seconds (a new config parameter).  Once min_uptime seconds elapse, the daemon may shut the system down, but this gives the user a grace period to log in and disable the service if we're actually running from USB.

Another option would be to *permanently* disable clean-shutdown (until restarted) if the very first poll of the GPIO indicates low battery (on the assumption we're actually not using a battery).  This would be really convenient when running from USB and avoid a new config parameter, but will behave badly if we're actually booting the system off a low battery.

I'd be happy to submit a pull request if this feature would be welcome.

Gadgetoid, 2017-07-18:
These are some good suggestions! I'll see what we can do. I'll summon @RogueM since this script is very much his baby.

RogueM, 2017-07-27:
from the README at the top of this repo:

> you may add the following to your /boot/config.txt file from another computer:
> `disable_cleanshutd=1`

RogueM, 2017-07-27:
incidentally, the `shutdown_delay` should already be 5 minutes? it's not exactly what you are asking, but as default for the Zero LIPO, at least if installed by passing the product parameter, I think it does end up having the same effect?

dbbnrl, 2017-07-28:
Huh.  Good point about shutdown_delay -- I agree it should be functionally equivalent to my suggestion.  Silly of me not to have considered it.  I need to go back and figure out why my device was shutting down almost instantly, since I didn't intentionally change that parameter.

Closing this issue since equivalent functionality should already exist.

closed 2017-07-28 by dbbnrl