pimoroni/clean-shutdown #8: Hyperpixel conflict?

Targett363, 2017-08-23:
First time I've done this, I've a Hyperpixel and Lipo shim but the Hyperpixel stops functioning with cleanshutd enabled. I'm assuming this is a conflict at BCM 2,3,4 or 14?
The Hyperpixel works up to the splash screen then goes blank.

Gadgetoid, 2017-08-23:
https://pinout.xyz/pinout/hyperpixel
https://pinout.xyz/pinout/zero_lipo

Looks like Zero LiPo shares its Battery Low pin with BCM4. Presumably the "goes blank" you describe is the Pi simply shutting down because when the splash comes up- which possibly has blue in it? - BCM4 is asserted and the shutdown script issues a shutdown.

It's impossible to use any other HAT or pHAT with HyperPixel directly. You must use the software i2c bus (on BCM10 and 11) to drive an IO expander which in turn talks to external devices. A non-trivial endeavour!

Targett363, 2017-08-23:
Awesome I'd assumed that was what was happening, good to know I can still add i2c device to those pins, I'll post a workaround if I get it figured.
Guess I best start reading through the source!

closed 2017-11-20 by Gadgetoid