pimoroni/clean-shutdown #11: failed to start but running

janc1, 2017-12-14:
I am using lipo shim.
I did used pimoroni dashboard on last rasbian OS to install this soft.

I am getting error message in journal
[image]

so I did check running processes 

[image]

strange, it is running, and when my batter get low it is switch off my raspberry.

I did check systemd service and there is also mentioned error message
[image]

I am not sure why there are two services with very similar name. I think it is mistake
cleanshud.service and cleanshutd.service
see last screenshot with red underline

Gadgetoid, 2018-03-27:
Sorry for the radio silence, did you ever get anywhere with this?

I'm currently working on a Python replacement for cleanshutd that works in new features and switches to low-level polling so that it's more responsive. One of the things I'll be doing is switching to pure systemd (the current setup uses a typical sprawling init script) which should help make issues like this easier to hunt down, or eliminate them altogether!

janc1, 2018-03-27:
well, I did reinstall rasbian on my card and it is working now.
it is good to hear that you will switch to pure systemd ...