pimoroni/clean-shutdown #12: Any way to run a script when shutdown invoked?

alc6379, 2018-03-26:
I'm using clean-shutdown, and I'd like to send a message to IFTTT that notifies me when shutdown is imminent. 

Is there any hook into clean-shutdown where I can have an additional command run, like doing a post to a web service? I'm sure I can modify the script that is being run, but I didn't know if there was some modularity that allowed me to extend the script built-in.

Gadgetoid, 2018-03-26:
Right now there's no easy way to hook into the shutdown event and run something, but it sounds like a worthwhile feature. I'd like to rewrite this script in Python to bring in some other requested features, that would be a good opportunity to add this.

alc6379, 2018-03-26:
Thanks. I wound up just modifying cleanshutd to add the webhook call via cURL. It was straightforward enough, but some type of directory where you could drop ad hoc scripts in, perhaps run in alphanumeric order (01-*.sh, 02-*.sh, etc) would be pretty neat.

Gadgetoid, 2018-10-15:
It strikes me, in retrospect, that this would be recreating functionality already supplied by systemd using `Before=shutdown.target` and we'd just be muddying the water by creating yet-another-way.

closed 2018-10-15 by Gadgetoid