pimoroni/clean-shutdown #23: one line installer not working

kidkill, 2019-05-26:
When I try to run (curl https://get.pimoroni.com/cleanshutdown | bash) , I get an error saying 

> Checking environment...
We can't connect to the Internet, check your network!

I definitely have an internet connection. Any help would be appreciated.

Gadgetoid, 2019-06-12:
This is a quirk in our connectivity check- we're pinging (last I knew) Google's DNS servers to verify an internet connection. Honeslty I think we just need to drop this check altogether and check exit codes for connection failures. It's been on my TODO list for a while.

Gadgetoid, 2019-11-06:
Have opened https://github.com/pimoroni/get/issues/202 to track this, since it applies across all of our "get" scripts.

closed 2019-11-06 by Gadgetoid