pimoroni/clean-shutdown #27: Possible to add button push function/script while the pi is on?

MattInSpace, 2020-10-15:
At the moment if the button is held for 1second the pi shuts down, which is great. Love that, exactly what I need. I'd really like to add something that issues a PiWakeup/CEC wakeup command over HDMI if the button is pushed but not held while the pi is already on. 

At the moment the Pi just goes to sleep and I'd like to wake it back up, rather than restarting it using the button. I could probably do this with a second button easily enough but would like to keep the hardware minimal and try to use the same button as on the On/Off Shim....

Gadgetoid, 2020-10-15:
Is there a command-line command you can use for wakeup?

Right now the existing shutdown script doesn't have any good way to detect a momentary button press- it relies upon the button being held down for its very lazy polling to detect it at all.

It would be pretty easy to rewrite this in Python and support what you want, though.

cluelesscarter, 2020-10-16:
@Gadgetoid, 

The command line `echo 'on 0' | cec-client -s` would wake it up over HDMI CEC using the `cec-utils`

Maybe a 0.5-second press could wake up and a ~3-second press could shut down? Or something like that?

New to this but happy to explore Python! Where would I 'add' it to the existing code?

MattInSpace, 2020-10-18:
@Gadgetoid  hey - sorry for the confusion using two accounts, one is work one is home. forgot which I was logged into. so the pi isn't going to sleep it's a function of MagicMirror, my bad. I've found some modules for the mirror which will allow me to cleanshut down using a button, so I guess new question - is it possible to disable the shutdown part of the ON/OFF shim code and leave the startup part running? 

I can shutdown and sleep using a magicmirror module, but not startutp.

Gadgetoid, 2020-10-21:
You would need the poweroff part of this script- I believe- to switch the latch in ON/OFF shim - https://github.com/pimoroni/clean-shutdown/blob/master/daemon/lib/systemd/system-shutdown/gpio-poweroff

Without this your system will shut down, but power will never be cut so the switch to restore power does nothing.

Just a case of copying that file into the right directory on your Pi:

```
sudo cp ./daemon/lib/systemd/system-shutdown/gpio-poweroff /lib/systemd/system-shutdown/gpio-poweroff
```

closed 2023-01-20 by Gadgetoid