pimoroni/clean-shutdown #28: init.d typo prohibits easy systemctl tabbing

rimeraz, 2020-12-17:
I tried to tab-complete after typing `systemctl status clea` and it completed only until `cleanshu`. Double-tab resulted in:
```
root@myhost:~# systemctl status cleanshu
cleanshud.service   cleanshutd.service
root@myhost:~# 
```

`cleanshud.service` is nowhere defined. It results from the typo in line 3 of `daemon/etc/init.d/cleanshutd`:

```
#! /bin/sh
### BEGIN INIT INFO
# Provides:          cleanshud
```

Actually it should read:

```
#! /bin/sh
### BEGIN INIT INFO
# Provides:          cleanshutd
```

After adding the missing `t` and `systemctl daemon-reload` the `systemctl status clea` will tab-complete as expected.

pirateradiohack, 2021-05-04:
I can see that Systemd creates 2 services as well, `cleanshud.service` and `cleanshutd.service`. 

I have created the pull request https://github.com/pimoroni/clean-shutdown/pull/30 in order to fix it.

Gadgetoid, 2021-05-04:
D'oh! Thank you.

closed 2021-05-04 by Gadgetoid