pimoroni/clean-shutdown #37: Pi does not shutdown any more after upgrade to bookworm

elhennig, 2024-02-15:
Hi, 
I upgraded my raspberry pi 4 to bookworm and kernel 6.6 and even if the GPIO pin 17 is triggered the clean shutdown down does not respond or shutdown the raspberry pi anymore.
I also changed the command from raspi-gpio to pinctrl as messages lead to the info the the command raspi-gpio is deprecated.
Any idea how to debug this more or even better solve the issue?

elhennig, 2024-02-15:
Strange is, that is was working before the OS and kernel update. I will check /boot/firmware/config.txt anyway.

elhennig, 2024-02-16:
I did some more testing: Shutdown works if I set the PIN 4 explicitly on the shell:
raspi-gpio set 4 op dl

The clean shutdown daemon confiuration look slike this:
`trigger_pin=17
led_pin=17
poweroff_pin=4
hold_time=1
shutdown_delay=0
polling_rate=1
`
However if I press the switch the PIN 4 is not changed, only 17. 

What is the correct setup: PIN 4 for shutdown and 17 for the restart trigger?

Gadgetoid, 2024-02-16:
That looks right-

* `trigger_pin` is the external switch
* `poweroff_pin` is the IO pin that's asserted at the very end of a shutdown to cut the power

If `raspi-gpio set 4 op dl` immediately cuts power, then you'll need to add `dtoverlay=gpio-poweroff,gpiopin=4,active_low=1` to `/boot/firmware/config.txt`.

There's also a straight dtoverlay approach for the shutdown button, but it doesn't have any long-press or LED indicator functionality: https://github.com/raspberrypi/linux/blob/rpi-6.1.y/arch/arm/boot/dts/overlays/gpio-shutdown-overlay.dts

And from the docs:
```
Name:   gpio-shutdown
Info:   Initiates a shutdown when GPIO pin changes. The given GPIO pin
        is configured as an input key that generates KEY_POWER events.

        This event is handled by systemd-logind by initiating a
        shutdown. Systemd versions older than 225 need an udev rule
        enable listening to the input device:

                ACTION!="REMOVE", SUBSYSTEM=="input", KERNEL=="event*", \
                        SUBSYSTEMS=="platform", DRIVERS=="gpio-keys", \
                        ATTRS{keys}=="116", TAG+="power-switch"

        Alternatively this event can be handled also on systems without
        systemd, just by traditional SysV init daemon. KEY_POWER event
        (keycode 116) needs to be mapped to KeyboardSignal on console
        and then kb::kbrequest inittab action which is triggered by
        KeyboardSignal from console can be configured to issue system
        shutdown. Steps for this configuration are:

            Add following lines to the /etc/console-setup/remap.inc file:

                # Key Power as special keypress
                keycode 116 = KeyboardSignal

            Then add following lines to /etc/inittab file:

                # Action on special keypress (Key Power)
                kb::kbrequest:/sbin/shutdown -t1 -a -h -P now

            And finally reload configuration by calling following commands:

                # dpkg-reconfigure console-setup
                # service console-setup reload
                # init q

        This overlay only handles shutdown. After shutdown, the system
        can be powered up again by driving GPIO3 low. The default
        configuration uses GPIO3 with a pullup, so if you connect a
        button between GPIO3 and GND (pin 5 and 6 on the 40-pin header),
        you get a shutdown and power-up button. Please note that
        Raspberry Pi 1 Model B rev 1 uses GPIO1 instead of GPIO3.
Load:   dtoverlay=gpio-shutdown,<param>=<val>
Params: gpio_pin                GPIO pin to trigger on (default 3)
                                For Raspberry Pi 1 Model B rev 1 set this
                                explicitly to value 1, e.g.:

                                    dtoverlay=gpio-shutdown,gpio_pin=1

        active_low              When this is 1 (active low), a falling
                                edge generates a key down event and a
                                rising edge generates a key up event.
                                When this is 0 (active high), this is
                                reversed. The default is 1 (active low).

        gpio_pull               Desired pull-up/down state (off, down, up)
                                Default is "up".

                                Note that the default pin (GPIO3) has an
                                external pullup. Same applies for GPIO1
                                on Raspberry Pi 1 Model B rev 1.

        debounce                Specify the debounce interval in milliseconds
                                (default 100)
                                
```

```
Name:   gpio-poweroff
Info:   Drives a GPIO high or low on poweroff (including halt). Using this
        overlay interferes with the normal power-down sequence, preventing the
        kernel from resetting the SoC (a necessary step in a normal power-off
        or reboot). This also disables the ability to trigger a boot by driving
        GPIO3 low.

        The GPIO starts in an inactive state. At poweroff time it is driven
        active for 100ms, then inactive for 100ms, then active again. It is
        safe to remove the power at any point after the initial activation of
        the GPIO.

        Users of this overlay are required to provide an external mechanism to
        switch off the power supply when signalled - failure to do so results
        in a kernel BUG, increased power consumption and undefined behaviour.
Load:   dtoverlay=gpio-poweroff,<param>=<val>
Params: gpiopin                 GPIO for signalling (default 26)

        active_low              Set if the power control device requires a
                                high->low transition to trigger a power-down.
                                Note that this will require the support of a
                                custom dt-blob.bin to prevent a power-down
                                during the boot process, and that a reboot
                                will also cause the pin to go low.
        input                   Set if the gpio pin should be configured as
                                an input.
        export                  Set to export the configured pin to sysfs
        active_delay_ms         Initial GPIO active period (default 100)
        inactive_delay_ms       Subsequent GPIO inactive period (default 100)
        timeout_ms              Specify (in ms) how long the kernel waits for
                                power-down before issuing a WARN (default 3000).
```

elhennig, 2024-02-16:
The straight power off is not really what I want. The question is then, why the shutdown daemon does not react on the PIN 17 lo signal. How can I debug the daemon script?

elhennig, 2024-02-16:
I found several problems with the bash script itself, with the outputs of raspi-gpio, that seem to have changed and some number comparison errors in if statements.
Still I do have an error after fixing the script: The systems shuts down but comes up again immediately (at least for the bootloader, the red LED is on, the OS not available

elhennig, 2024-02-18:
I am using the OnOffShim.

Do I understand right, that the dtoverlay triggers a real shutdown like the command "shutdown -h". What is the better overlay gpio-shutdown or gpio-poweroff in my case? In bullseye the pi completely shut down (red LED off). 
Can this be done with the overlay config?

elhennig, 2024-02-18:
I use 
`dtoverlay=gpio-shutdown,gpio_pin=17,active_low=1,gpio_pull=up`
and it works sufficiently good. Is it right, that I cannot trigger a startup without having cycled the power before?

Gadgetoid, 2024-02-19:
You need to use *both* `gpio-shutdown` which- and I may mangle terminology here- just acts like a real ACPI power button *and* `gpio-poweroff` which toggles a pin to - hopefully - trigger your external board to cut the power completely.

OnOffShim, IIRC, then can latch the power back on when you press the button again.

So maybe:

```
dtoverlay=gpio-shutdown,gpio_pin=17,active_low=1,gpio_pull=up
dtoverlay=gpio-poweroff,gpiopin=4,active_low=1
```

Then OnOffShim will cycle the power for you. (Though *only* if you have the power connected via OnOffShim, if you use the Pi's power connector it can't work)

Ramblurr, 2024-06-24:
@elhennig could you share the working config you ended up with?

ikedah-beyond, 2024-07-14:
Hello, excuse my abrupt writing.
I had a similar problem on my Pi3 and OnOff-SHIM.

I think one of the causes of the trouble is that the number given to /sys/class/gpio was changed by bookwarm, and the other is that the format of the return value of raspi-gpio get was changed.
So, it seems that the following two modifications in the code avoided the trouble.

/lib/systemd/system-shutdown/gpio-poweroff

# change gpio num for bookworm
#cat /sys/kernel/debug/gpio | grep GPIO$poweroff_pin |cut -f2 -d ' ' |cut -f2 -d '-'
poweroff_pin=516
#cat /sys/kernel/debug/gpio | grep GPIO$led_pin |cut -f2 -d ' ' |cut -f2 -d '-'
led_pin=529 

/user/bin/cleanshutd

trigger_pin_low() {
   # raspi-gpio get $trigger_pin | grep -q "level=0 fsel=0 func=INPUT"
   raspi-gpio get $trigger_pin | grep -q "level=0 func=INPUT"
}

Please refer to the following.

rodubero, 2024-08-20:
Hi. The original installation worked in RPi4 with raspios_arm64-2024-07-04 after:

- Using `/boot/firmware/config.txt` instead of `/boot/config.txt` (i.e in CONFIG var of setup.sh or just relocating config.txt) 
- [Adding the dtoverlays](https://github.com/pimoroni/clean-shutdown/issues/37#issuecomment-1952129162) recommended by  @Gadgetoid in `/boot/firmware/config.txt`
- Replacing all the references of `raspi-gpio` in `/usr/bin/cleanshutd` for `pinctrl`

Haven't checked more details yet about the possible changes in setup.sh, but it was the quick and dirty way to fix :)

Barbs56, 2025-02-12:
There is a lot of disparate info here... What is the exact solution per the manufacturer at this point? The only thing I've seen is adding dtoverlays to /boot/firmware/config.txt along with hotfixes (which don't seem to line up between the posts above). Thanks in advance.

Ramblurr, 2025-02-13:
> There is a lot of disparate info here... What is the exact solution per the manufacturer at this point? The only thing I've seen is adding dtoverlays to /boot/firmware/config.txt along with hotfixes (which don't seem to line up between the posts above). Thanks in advance.

@Gadgetoid is a commiter to this repo, so that is about as close to an answer from the vendor as you are going to get. Other than that pimoroni has abandoned maintenance of this product.

However as I noted in my comment, @Gadgetoid 's recommendation does not provide the same features as the original implementation (i.e., a clean shutdown with systemd). I've been using the technique in my comment for nearly a year without problems. If you understand systemd and understand the `gpio-poweroff` and `gpio-shutdown` docs, then crafting your own solution is fairly simple.

Barbs56, 2025-02-13:
@Ramblurr thanks, your response seemed the most complete. However, I was unclear on exactly what you did. Did you execute the one line installer, then add /usr/lib/systemd/system-shutdown/gpio-shutoff /usr/local/bin/onoff-shim-trigger /etc/systemd/system/onoff-trigger.service, in addition to adjusting the overlays? Or, does your approach assume the Pimoroni installer was not executed? I've already run the installer, so, if it is causing conflicts, I'll have to figure out how to track down the files it added and remove them. I'm not a Linux pro, so the simple installation was a huge reason I bought this device.

rodubero, 2025-02-13:
@Barbs56 try [this fork](https://github.com/rodubero/clean-shutdown) :)

Barbs56, 2025-02-13:
@rodubero THANK YOU. This seemed to work instantly. One question: I already ran `curl https://get.pimoroni.com/onoffshim | bash` when I first installed the switch. Do the instructions in the fork you provided overwrite all files or do I need to clean anything up from the first install try?

Ramblurr, 2025-02-14:
> [@Barbs56](https://github.com/Barbs56) try [this fork](https://github.com/rodubero/clean-shutdown) :)

Ah nice job making a fork for the community! I wish it wasn't necessary, but nice that someone has done it.